adafruit/Adafruit_CircuitPython_AM2320 #3: fails intermittently on raspberry pi

brennen, 2018-08-10:
Testing on Pi with latest version of the library.  Reads roughly every other attempt.

Tried with 10k and 1k resistors and with none at all, no difference.  Not sure if I'm missing something there.

```
$ python am2320.py 
Humidity: 46.8%
Temperature: 25.4C
Traceback (most recent call last):
  File "am2320.py", line 10, in <module>
    print('Humidity: {0}%'.format(sensor.relative_humidity))
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/adafruit_am2320.py", line 125, in relative_humidity
    humidity = struct.unpack(">H", self._read_register(AM2320_REG_HUM_H, 2))[0]
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/adafruit_am2320.py", line 93, in _read_register
    i2c.write(bytes([0x00]))
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/adafruit_bus_device/i2c_device.py", line 102, in write
    self.i2c.writeto(self.device_address, buf, **kwargs)
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/busio.py", line 61, in writeto
    return self._i2c.writeto(address, buffer, stop)
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/adafruit_blinka/microcontroller/raspi_23/i2c.py", line 36, in writeto
    self._i2c_bus.write_bytes(address, buffer)
  File "/home/pi/blinkenlight/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 244, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error
```

Checking with `i2cdetect` twice in a row:

```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

cc: @kattni, @ladyada

jerryneedell, 2018-08-10:
The AM2320 has some "odd" wake-up behavior that mey be relevant.  see the discussion at 
 https://github.com/adafruit/circuitpython/issues/1053

ladyada, 2018-10-02:
added retries - the i2cscan will alway sbe flakey, you have to run it twice,  but at least the example doesnt need to be run twice. wanna do this one next once my PR is in?

closed 2018-10-02 by caternuson