adafruit/Adafruit_CircuitPython_AM2320 #4: Takes two reads to get current sensor value.

caternuson, 2018-10-01:
This is very likely related to #3. The first request wakes it up. The second request actually returns the value. Maybe?
```python
Adafruit CircuitPython 3.0.2 on 2018-09-14; Adafruit Metro M4 Express with samd51j19
>>> import board, busio
>>> import adafruit_am2320
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> sensor = adafruit_am2320.AM2320(i2c)
>>> sensor.relative_humidity
53.8
>>> # use breath to generate some humidity
>>> sensor.relative_humidity
53.8
>>> sensor.relative_humidity
99.6
>>> # do nothing for ~1 minute
>>> sensor.relative_humidity
99.8
>>> sensor.relative_humidity
58.3
>>> 
```

caternuson, 2018-10-02:
So....this is probably just a feature of this inexpensive sensor. Reading twice seems to be the inexpensive solution.

closed 2018-10-02 by caternuson

ladyada, 2018-10-02:
more specifically - reading heats it up a bit, which will evaporate condensation! so more reading = faster 'cool down'