adafruit/Adafruit_CircuitPython_AM2320 #8: Use locally defined exception classes instead of raising generic exception types.

Ferroin, 2019-10-08:
Prompted by: https://github.com/netdata/netdata/pull/7024

This module appears to directly raise `ValueError` and `RuntimeError` under specific circumstances.  While both are technically accurate, they necessitate use of overly broad `except` clauses when dealing with error handling for the module, which makes it difficult to verify that code using the module is actually correct and can't accidentally hide other errors. The `RuntimeError` usage is particularly problematic, as it's very rare in the core language to ever catch a `RuntimeError`, so any code that catches one tends to look rather out of place to many people.

I would suggest switching to the same methodology that's used by much of the Python ecosystem of defining local sub-classes of `Exception` (or whatever specific error makes sense) and raising those instead of directly raising built-in exception types.

ladyada, 2019-10-08:
hiya, if you'd like to submit a PR for an update, we'd love to have one - note that this code must be able to run on CircuitPython boards that have very limited RAM

tannewt, 2019-10-08:
@Ferroin That would be awesome to add! Let me know if you have any questions on how to do it or test it.

Ferroin, 2019-10-08:
@tannewt Testing should be the only issue (I'm a bit out of practice with Python in general, but that shouldn't be a problem for something like this).  I've never worked with CircuitPython myself (as mentioned in the issue description, this was prompted by a PR for Netdata to add a data collector that uses this module), so I'm not sure where exactly the best place to begin would be.

tannewt, 2019-10-08:
Here is a guide on how to use the sensor: https://learn.adafruit.com/adafruit-am2320-temperature-humidity-i2c-sensor/python-circuitpython

Adding the custom exceptions should be the same as CPython so it's ok to only test in CPython with Blinka.

PR #9 (2019-12-31): Add custom exception classes instead of using plain error types.
This adds custom exception classes so that users do not have to match on generic exception types in code that uses this module.

The new custom classes are structured so that existing code that is matching on the generic exception types will still work, and so that there's a primary exception type that can be matched on to catch all
errors specific to this module.

----

Fixes: #8 

This is minimally tested right now (using `flake8` to check that there are no syntax errors or other obvious problems with the changes). I intend to  [... cut]
closed 2019-12-31 by ladyada