adafruit/Adafruit_CircuitPython_AM2320 #15: Custom Exceptions Causing TypeErrors

sommersoft, 2020-04-14:
A user in Discord brought this issue up. They receive the following exception when trying to use the library:
```
File "adafruit_am2320.py", line 84, in <module>
TypeError: multiple bases have instance lay-out conflict
```

[Line 84](https://github.com/adafruit/Adafruit_CircuitPython_AM2320/blob/3a0867e9d72340de6d9ae7bfa8c5659b35c0796c/adafruit_am2320.py#L84) points to a custom exception, which came from #9. This was included in release `1.1.4` onward.

The user confirmed that release `1.1.3` does not raise the TypeError exception.

jerryneedell, 2020-04-16:
hmm -- I reproduced this error on a few CP boards -- grandcentral, teensy 4.0

but it works fine an a Raspberry Pi 4...

jepler, 2020-04-16:
Yes, it looks like this construct is not supported on CircuitPython.  We're not likely to be able to add support for it, so unfortunately I don't see a way forward except for reverting it.  Perhaps someone else does.

```
>>> class E(Exception): pass
... 
>>> class D(E, ValueError): pass
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: multiple bases have instance lay-out conflict
```

Any time an inheritance tree of classes includes more than one native base class among all its bases, this error will occur.

jerryneedell, 2020-04-16:
I  must say that this is one of most perplexing error messages I have ever received ```TypeError: multiple bases have instance lay-out conflict```  mostly due to my own ignorance of Python details, but still, could it be a bit more obtuse? ;-)

tannewt, 2020-04-21:
Could we just have the custom Exceptions not inherit from two types? Specifically, I think we can remove AM2320Exception and just leave the stock parents.

jerryneedell, 2020-04-21:
> Could we just have the custom Exceptions not inherit from two types? Specifically, I think we can remove AM2320Exception and just leave the stock parents.

I have no idea what that means, but it sounds fine to me ;-)
I only jumped in because it was suggested that it be reverted and it was simple to do.
If you can tell me what to change, I can update the PR or feel free to take control  of this.

jerryneedell, 2020-04-21:
@tannewt  ah -- I see the PR was merged -- Do you want to create a ne PR to address this?  Should I hold off on releasing?

tannewt, 2020-04-22:
@jerryneedell I don't plan on making a PR but it'd be great if someone else would. I think it's ok to do a major release now to get the fix out.

jposada202020, 2021-04-23:
Closing. Thanks.

closed 2021-04-23 by jposada202020