adafruit/Adafruit_CircuitPython_AM2320 #17: I2C read failure on AM2315 sensor

lucabar, 2020-05-09:
Hello everyone,

I am currently trying to use an AM2315 temperature and humidity sensor on a Raspberry Pi 4 and am running into an I2C read failure. In the <a href= "https://cdn-learn.adafruit.com/downloads/pdf/am2315-encased-i2c-temperature-humidity-sensor.pdf?timestamp=1589040136"> PDF that came with the sensor</a> it is stated that the AM2320 driver works just as well for the AM2315. And since I could not find any other maintained python driver for the AM2315, I had to use the AM2320 driver as I was told. So now when I run a simple program (similar to your "am2320_simpletest.py") 
<pre>
<code>import board, busio, adafruit_am2320, time
i2c = busio.I2C(board.SCL, board.SDA)
my_sens = adafruit_am2320.AM2320(i2c)
while True:
	print(my_sens.temperature,my_sens.relative_humidity)
	print('')
	time.sleep(0.5)</code>
</pre>
it returns the humidity and temperature fine for two runs
<code>21.1 44.1</code>
<code>21.5 44.3</code>
but then returns
<pre>Traceback (most recent call last):
  File "am2320_try.py", line 11, in <module>
    print(my_sens.temperature)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 144, in temperature
    temperature = struct.unpack(">H", self._read_register(AM2320_REG_TEMP_H, 2))[0]
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 133, in _read_register
    raise AM2320ReadError("I2C read failure")
adafruit_am2320.AM2320ReadError: I2C read failure</pre>
This happens every time after two measurements (or less, when I increase time.sleep()) were conducted and I guess it has something to do with the fact that "preamble indicates correct readings" (see adafruit_am2320.py, line 118) but I can not really make anything of it.
Can you help me in this regard? Maybe if it's an issue that I'm working with AM2315 instead of AM2320?

Thanks a lot in advance!

ladyada, 2020-05-09:
yeah you probably cannot use AM2315 with AM2320 library

dherrada, 2020-06-17:
@ladyada Should I edit the learn guide so it no longer says this or would you like me to try to work on a fix? Alternatively, since this library did work with the AM2315 at some point within the last two years, I bet it wouldn't be too hard to turn this into AM23XX and have it support for both of them in separate classes

ladyada, 2020-06-28:
@dherrada hihi i cant find where you see us saying AM2315 works with AM2320?

dherrada, 2020-06-29:
@ladyada https://learn.adafruit.com/am2315-encased-i2c-temperature-humidity-sensor/python-circuitpython

ladyada, 2020-06-29:
oh wait now i remember the issue - you cant query it too often.
@lucabar try adding a time.sleep(2)

lucabar, 2020-06-30:
I've tried that already. The only thing that changes is that the read-failure comes right after the first measurement of temperature and humidity (and not after the second as above).

ladyada, 2020-06-30:
if you catch the exception and try again, does it work later?

beitong95, 2020-07-09:
[image]
This works for me. I think we should write a function to query temp and humidity together.

dherrada, 2020-08-06:
> I think we should write a function to query temp and humidity together.

The issue is that you can't query the AM2320 too often. If we were to do a function that got both of them, it'd need to query it twice since the temperature and humidity are in different registers.

dherrada, 2020-08-06:
@lucabar Have you tried catching the exception like @ladyada suggested? I've encountered similar issues with other libraries, and that is sometimes the only way to get it to work right.

meteorkees, 2020-11-27:
Hello,
since a week ore soo I use the AM2315 sensor to read the outside temp and RV. I send the feeds to IO Adafruit with the following code:

`import time
import board
import busio
import adafruit_am2320 #am2315 works with am2320 lib
# import Adafruit IO REST client.
from Adafruit_IO import Client, Feed

AM2315_READ_TIMEOUT = 5

# Set to your Adafruit IO key.
# Remember, your key is a secret,
# so make sure not to publish it when you publish this code!
ADAFRUIT_IO_KEY = 'key'

# Set to your Adafruit IO username.
# (go to https://accounts.adafruit.com to find your username).
ADAFRUIT_IO_USERNAME = 'username'

# Create an instance of the REST client.
aio = Client(ADAFRUIT_IO_USERNAME, ADAFRUIT_IO_KEY)

# Set up Adafruit IO Feeds.
#outside_temperature_feed = aio.feeds('outside_temperature')
#outside_humidity_feed = aio.feeds('outside_humidity')

# create the I2C shared bus
i2c = busio.I2C(board.SCL, board.SDA)
am = adafruit_am2320.AM2320(i2c)
     
while True:
    outside_temperature = '%.1f'%(am.temperature)
    outside_relative_humidity = '%.1f'%(am.relative_humidity)
    if outside_relative_humidity is not None and outside_temperature is not None:
        print("outside_Temperature: ", am.temperature)
        print("outside_Humidity: ", am.relative_humidity)
        # Send humidity and temperature feeds to Adafruit IO
        aio.send('outsidetemperature', str(outside_temperature))
        aio.send('outsidehumidity', str(outside_relative_humidity))
        time.sleep(30)
    else:
        print('Failed to get AM2315 Reading, trying again in ', AM2315_READ_TIMEOUT, 'seconds')
    # Timeout to avoid flooding Adafruit IO
    time.sleep(AM2315_READ_TIMEOUT)`

But after a while (about 10-12 hours) I get this error in return:

Traceback (most recent call last):
  File "/home/pi/io-client-python/examples/basics/outside_T_RH.py", line 35, in <module>
    print("outside_Humidity: ", am.relative_humidity)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 137, in relative_humidity
    humidity = struct.unpack(">H", self._read_register(AM2320_REG_HUM_H, 2))[0]
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 118, in _read_register
    raise RuntimeError("I2C read failure")
RuntimeError: I2C read failure

Any idea or suggestions what could be wrong?

Thanks in advance and take care,

Kees

meteorkees, 2020-11-27:
I will find out how to paste the code. This looks no good :/

ladyada, 2020-11-28:
try catching the error, waiting for a while, and trying again for more tech support questions you can post in the forums

closed 2020-11-28 by ladyada

skyblack2, 2021-07-29:
https://github.com/switchdoclabs/SDL_Pi_AM2315 use this one is gonna be fine