adafruit/Adafruit_CircuitPython_AM2320 #18: strange discontinuity between 31 and 33 degrees C?

2bitoperations, 2020-06-04:
Howdy! 
I'm noticing a strange "jump" in temperature readings between 31 and 33 celcius.
[image]
[image]

The behavior is pretty consistent, and I am pretty sure the sensor is in error within this range. Has anyone else experienced this?

dherrada, 2020-06-17:
Hi @2bitoperations are you still experiencing this issue? I may be able to take a look at this in a few days if you are.

2bitoperations, 2020-07-01:
Yikes - sorry for the slow reply. Yes, still experiencing the issue.

amm0nite, 2020-07-01:
Hello, I had a similar problem with a lot of jumps between 23 and 24° C on my Raspberry Pi, I have a workaround on the software side by waiting 500ms and re doing the measurement, repeat until the difference with the previous temperature is under a 0.3° threshold. Usually it takes only one retry.