adafruit/Adafruit_CircuitPython_AM2320 #23: I2C read error after the first measure

B1ack70, 2021-11-01:
Hi,

After the first measure, I have this I2C read error :

```
Python 3.7.3 (/usr/bin/python3)
>>> %Run test_AM2320.py
Temperature:  21.9
Humidity:  58.7
Traceback (most recent call last):
  File "/home/pi/Documents/AM2320/test_AM2320.py", line 9, in <module>
    print("Temperature: ", am.temperature)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 138, in temperature
    temperature = struct.unpack(">H", self._read_register(AM2320_REG_TEMP_H, 2))[0]
  File "/usr/local/lib/python3.7/dist-packages/adafruit_am2320.py", line 127, in _read_register
    raise RuntimeError("I2C read failure")
RuntimeError: I2C read failure
```
The code is the github example : am2320_simpletest.py
I have tried to add more time between two measures but it's not working.
This issue looks the same of the number 17 (https://github.com/adafruit/Adafruit_CircuitPython_AM2320/issues/17) but I have a AM2320 sensor.

Any idea ? ;)
Thank's.

caternuson, 2022-01-31:
Can not recreate issue with a quick test on a Raspberry Pi 4:
```
(blinka) pi@raspberrypi:$ python3 am2320_simpletest.py 
Temperature:  22.3
Humidity:  36.7
Temperature:  21.9
Humidity:  36.7
Temperature:  21.6
Humidity:  36.7
Temperature:  21.7
Humidity:  36.7
Temperature:  21.6
Humidity:  36.7
Temperature:  21.6
Humidity:  36.6
Temperature:  21.7
Humidity:  36.7
```

caternuson, 2022-01-31:
Potentially related issue seen with Feather M4 and RP2040:
https://forums.adafruit.com/viewtopic.php?f=60&t=187866

caternuson, 2022-02-02:
Closing for now. Forum thread was using non-Adafruit sourced sensor. Can reopen if issue can be demonstrated with one of these:
https://www.adafruit.com/product/3721

closed 2022-02-02 by caternuson

B1ack70, 2022-02-02:
Thank's for your answer. I'm using a RPi 4B.