adafruit/Adafruit_CircuitPython_AM2320 #27: AM2315 OSError: [Errno 19] No such device

fabianmuehlberger, 2022-09-13:
```
Temperature:  25.9
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "/lib/adafruit_am2320.py", line 149, in relative_humidity
  File "/lib/adafruit_am2320.py", line 123, in _read_register
OSError: [Errno 19] No such device
```
The error occurs if temperature and humidity are read without delay between readings.

following code is used:
```
import board
import busio
import time
import adafruit_am2320

i2c = busio.I2C(scl=board.GP17, sda=board.GP16)
am = adafruit_am2320.AM2320(i2c)

while True:
    print("Temperature: ", am.temperature)
    time.sleep(0.03) 
    print("Humidity: ", am.relative_humidity)
    time.sleep(1)
```
at least 0.03 seconds of sleep is needed for the code to run correctly.

running on a raspberry pico W
circuitpyton v 8.0.0 
no other devices attached
10k resistors are used to decouple scl and sda
the sensor is supplied with 3.3 V

tekktrik, 2022-09-13:
@dhalbert @jepler were there any changes to I2C in CircuitPython 8.0?  The fix could be here to just add some more delay between readings, but wanted to check.

happy4pizza, 2023-03-27:
I believe the issue is caused because you are sending two data packets at the same time from a device with only one IC address, I think this is distorting the data so your microcontroller can't read it. I would just increase time.sleep. This seems to  be a hardware limitation and can only be fixed if we change the board.

tyeth, 2023-08-21:
Similar issue in arduino https://github.com/adafruit/Adafruit_AM2320/issues/13

closed 2024-11-06 by dhalbert